rpng/kalibr_allan #2: Which MATLAB script to run?

CtfChan, 2018-01-21:
Hi there, which MATLAB script am I supposed to run after converting the Rosbag to a .mat file?

goldbattle, 2018-01-21:
The parallel script should be there most up to date. I have not had time to
verify the correctness of the scripts yet.

CtfChan, 2018-01-21:
Thanks man! I'll give it a try.

goldbattle, 2018-01-21:
After it process the data, you should be able to run the results script that will graph and build the charts for you. The parallel script just computes the Allen deviation of the given data, and will export to a mat file.

CtfChan, 2018-01-27:
Hi goldbattle, if I have a 3Hr IMU file how long should it take to run the parallel script? I've been running it for a while and the MATLAB terminal still saids "calculating allan deviation"

closed 2018-02-04 by goldbattle