rpng/kalibr_allan #3: IMU data format

citeacecsn, 2018-02-24:
Thanks for your  allan tool. I have two quetion

1.Is the IMU data format  (timestamp(s **not ns**), acc_x(m/s^2), acc_y(m/s^2), acc_z(m/s^2) , gyro_x(rad/s), gyro_y(rad/s), gyro_z(rad/s) for allan tool ?
2.The tool will fit a -1/2(or 1/2) line to the left(or right) side of the allan plot, however the line is not tangent line. Is it right? 
Thank you again.

[image]

goldbattle, 2018-02-25:
Correct the IMU .mat generated has seconds in the first column.
https://github.com/rpng/kalibr_allan/blob/master/bagconvert/src/main.cpp#L78

The -1/2 and +1/2 slopes are just "fitted" to the left and right parts of the allan chart, it isn't the tanget line yes. The Kalibr wiki along with some of the references all talk about fitting a 1/2 slope line to get the noise values.

Note this tool has not be extensively tested, so if you get any reasonable results, please let me know.

citeacecsn, 2018-02-26:
Thanks again.
The acc unit is m/s^2 and the gyro unit is rad/s in your tool? right?
I used your tool to get  some results, however I think the white noise and random walk noise are large.
[image]
[image]

goldbattle, 2018-02-26:
The units should be in what your *sensor* is publishing, which for ROS is
m/s^2 and rad/s.
Yeah, most of the noise values that I remember getting where far larger
then what they have as examples in the Kalibr wiki.
Not sure if there is some underlying data issue, or the method is wrong.

citeacecsn, 2018-02-26:
Get it, thank you.

closed 2018-02-26 by citeacecsn