rpng/kalibr_allan #5: can't find MATLAB_xxx_LIBRARYs althou I have installed matlab

duyilong, 2018-04-09:
when I run `catkin_make` at catkin workspace, error happened and error message says  "can't find MATLAB_ENG_LIBRARY (ADVANCED)"and other several MATLAB-libraries.

> CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
MATLAB_ENG_LIBRARY (ADVANCED)
    linked by target "bagconvert" in directory /home/duyl/catkin_ws/src/kalibr_allan/bagconvert
    linked by target "bagconvert" in directory /home/duyl/catkin_ws/src/kalibr_allan/bagconvert
MATLAB_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /home/duyl/catkin_ws/src/kalibr_allan/bagconvert
   used as include directory in directory /home/duyl/catkin_ws/src/kalibr_allan/bagconvert
   used as include directory in directory /home/duyl/catkin_ws/src/kalibr_allan/bagconvert
   used as include directory in directory /home/duyl/catkin_ws/src/kalibr_allan/bagconvert
   used as include directory in directory /home/duyl/catkin_ws/src/kalibr_allan/bagconvert
   used as include directory in directory /home/duyl/catkin_ws/src/kalibr_allan/bagconvert
   used as include directory in directory /home/duyl/catkin_ws/src/kalibr_allan/bagconvert
MATLAB_MAT_LIBRARY (ADVANCED)
    linked by target "bagconvert" in directory /home/duyl/catkin_ws/src/kalibr_allan/bagconvert
MATLAB_MEX_LIBRARY (ADVANCED)
    linked by target "bagconvert" in directory /home/duyl/catkin_ws/src/kalibr_allan/bagconvert
    linked by target "bagconvert" in directory /home/duyl/catkin_ws/src/kalibr_allan/bagconvert
MATLAB_MX_LIBRARY (ADVANCED)
    linked by target "bagconvert" in directory /home/duyl/catkin_ws/src/kalibr_allan/bagconvert
    linked by target "bagconvert" in directory /home/duyl/catkin_ws/src/kalibr_allan/bagconvert
MATLAB_UT_LIBRARY
    linked by target "bagconvert" in directory /home/duyl/catkin_ws/src/kalibr_allan/bagconvert

I install matlab2014A to /home/MY_NAME/Downloads/matlab, cause there is no enough free space at/usr/local, does that matter?

duyilong, 2018-04-11:
I tried that, executable found, but still can't find other libs

duyilong, 2018-04-11:
@goldbattle  I reinstall matlab at /usr/local (installer default option),
neither 
> find_program(MATLAB_EXE_PATH matlab
            PATHS /usr/local/MATLAB/R2014a/bin)

or
> find_program(MATLAB_EXE_PATH matlab
            PATHS /usr/local/bin)

works, same error happens, excutable found, but MATLAB not found

goldbattle, 2018-04-23:
It might not be compatible with the version of MATLAB you are using.
I am using R2016b.
Let me know if you figure anything out, thanks!

raoulj, 2018-05-02:
I had similar messages and got catkin to build successfully (with a warning) with a quick and dirty fix.  I changed the library-finding chunk in `findMatlab.cmake` to:

```
...
# This is common to all platforms:
# Get paths to the Matlab MEX libraries
find_library(MATLAB_MEX_LIBRARY
        libmex.so
        HINTS /usr/local/bin/matlab/bin/glnxa64/
)
find_library(MATLAB_MX_LIBRARY
        libmx.so
        HINTS /usr/local/bin/matlab/bin/glnxa64/
)
find_library(MATLAB_MAT_LIBRARY
        libmat.so
        HINTS /usr/local/bin/matlab/bin/glnxa64/
)
find_library(MATLAB_ENG_LIBRARY
        libeng.so
        HINTS /usr/local/bin/matlab/bin/glnxa64/
)
find_library(MATLAB_UT_LIBRARY
        libut.so
        HINTS /usr/local/bin/matlab/bin/glnxa64/
)

# Get path to the include directory
find_path(MATLAB_INCLUDE_DIR
        "mex.h"
        HINTS  "/usr/local/bin/matlab/extern/include"
)
...
```

The warning was related to `libboost_system` and `libboost_filesystem` having conflicting version needs.

nickcharron, 2018-12-14:
For those who are having this issue, here's how I solved it. 

I had the exact same problems as @duyilong, and when making the changes suggested by @goldbattle, it also didn't seem to fix the problem. But after deleting the build and devel directories and rebuilding, it worked fine! Hope this helps.

closed 2019-02-13 by goldbattle

rrison, 2019-07-09:
> For those who are having this issue, here's how I solved it.
> 
> I had the exact same problems as @duyilong, and when making the changes suggested by @goldbattle, it also didn't seem to fix the problem. But after deleting the build and devel directories and rebuilding, it worked fine! Hope this helps.

Great! My Matlab is installed in /usr/local/MATLAB/R2018a/bin, and I replace
    find_program(MATLAB_EXE_PATH matlab
            PATHS /usr/local/bin)
to
    find_program(MATLAB_EXE_PATH matlab
            PATHS /usr/local/MATLAB/R2018a/bin)
then rm -rf build/ && catkin_make
it works.

suho0515, 2020-04-23:
> Great! My Matlab is installed in /usr/local/MATLAB/R2018a/bin, and I replace
>
> ```
>     find_program(MATLAB_EXE_PATH matlab
>             PATHS /usr/local/bin)
> ```
>
> to
>
> ``` [...]

it sure worked for me, thanks!

FaboNo, 2020-09-21:
Dear All

Following your instruction, looks like it works for R2018 but unfortunately, it failed for the R2020 version :(

```
/usr/local/MATLAB/R2020b/bin/glnxa64/libmwservices.so: undefined reference to `__cxa_init_primary_exception@CXXABI_1.3.11'
/usr/local/MATLAB/R2020b/bin/glnxa64/libmwservices.so: undefined reference to `std::thread::_State::~_State()@GLIBCXX_3.4.22'
/usr/local/MATLAB/R2020b/bin/glnxa64/libmwservices.so: undefined reference to `std::thread::_M_start_thread(std::unique_ptr<std::thread::_State, std::default_delete<std::thread::_State> >, void (*)())@GLIBCXX_3.4.22'
/usr/local/MATLAB/R2020b/bin/glnxa64/libmwservices.so: undefined reference to `std::__exception_ptr::exception_ptr::exception_ptr(void*)@CXXABI_1.3.11'
/usr/local/MATLAB/R2020b/bin/glnxa64/libmex.so: undefined reference to `std::uncaught_exceptions()@GLIBCXX_3.4.22'
/usr/local/MATLAB/R2020b/bin/glnxa64/libmwservices.so: undefined reference to `typeinfo for std::thread::_State@GLIBCXX_3.4.22'
/usr/local/MATLAB/R2020b/bin/glnxa64/libmx.so: undefined reference to `operator delete(void*, unsigned long, std::align_val_t)@CXXABI_1.3.11'
/usr/local/MATLAB/R2020b/bin/glnxa64/libmx.so: undefined reference to `operator new(unsigned long, std::align_val_t)@CXXABI_1.3.11'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/fabrice/catkin_allan2_ws/devel/.private/bagconvert/lib/bagconvert/bagconvert] Error 1
make[1]: *** [CMakeFiles/bagconvert.dir/all] Error 2
make: *** [all] Error 2
```
Any hints? Or do I have to look for a R2018 version somewhere?

Thanks for the great works!

guchangjun, 2021-04-21:
I delete the findmatlab.cmake and moify the cmakelist as :

cmake_minimum_required(VERSION 2.8.3)
# Project name
project(bagconvert)
# Find catkin (the ROS build system)
find_package(catkin REQUIRED COMPONENTS roscpp nav_msgs std_msgs sensor_msgs nav_msgs rosbag)

# Describe catkin Project
catkin_package(
        DEPENDS Eigen3 Boost
        CATKIN_DEPENDS roscpp nav_msgs std_msgs sensor_msgs nav_msgs rosbag
        INCLUDE_DIRS src
)
# Include our cmake files
SET(CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake/)

# Include libraries
find_package(Eigen3 REQUIRED)
find_package(Boost REQUIRED COMPONENTS system filesystem thread date_time)
#find_package(Matlab 3 REQUIRED)

set(MATLAB_INCLUDE_DIR "/usr/local/MATLAB/R2018a/extern/include")
set(MATLAB_LIBRARIES_DIR "/usr/local/MATLAB/R2018a/bin/glnxa64")
# Include our header files
include_directories(
        src
        ${EIGEN3_INCLUDE_DIR}
        ${Boost_INCLUDE_DIRS}
        ${MATLAB_INCLUDE_DIR}
        ${catkin_INCLUDE_DIRS}
)
##################################################
# Make binary for the offline reader
##################################################
add_executable(bagconvert
        src/main.cpp
)
target_link_libraries(bagconvert 
        ${Boost_LIBRARIES}
#        ${MATLAB_LIBRARIES}
        ${MATLAB_LIBRARIES_DIR}/libmex.so
        ${MATLAB_LIBRARIES_DIR}/libmat.so
        ${MATLAB_LIBRARIES_DIR}/libeng.so
        ${MATLAB_LIBRARIES_DIR}/libmx.so
#        ${MATLAB_MX_LIBRARY}
#        ${MATLAB_MEX_LIBRARY}
#        ${MATLAB_MAT_LIBRARY}
       # ${MATLAB_ENG_LIBRARY}
        ${catkin_LIBRARIES})

Finally, it worked fine! Hope this helps

Yngchu, 2021-12-07:
> I delete the findmatlab.cmake and moify the cmakelist as :
> 
> cmake_minimum_required(VERSION 2.8.3)
> 
> # Project name
> project(bagconvert)
> 
> # Find catkin (the ROS build system)
> find_package(catkin REQUIRED COMPONENTS roscpp nav_msgs std_msgs sensor_msgs nav_msgs rosbag)
> 
> # Describe catkin Project
> catkin_package( DEPENDS Eigen3 Boost CATKIN_DEPENDS roscpp nav_msgs std_msgs sensor_msgs nav_msgs rosbag INCLUDE_DIRS src )
> 
> # Include our cmake files
> SET(CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake/)
> 
> # Include libraries
> find_package(Eigen3 REQUIRED) find_package(Boost REQUIRED COMPONENTS system filesystem thread date_time) #find_package(Matlab 3 REQUIRED)
> 
> set(MATLAB_INCLUDE_DIR "/usr/local/MATLAB/R2018a/extern/include") set(MATLAB_LIBRARIES_DIR "/usr/local/MATLAB/R2018a/bin/glnxa64")
> 
> # Include our header files
> include_directories( src ${EIGEN3_INCLUDE_DIR} ${Boost_INCLUDE_DIRS} ${MATLAB_INCLUDE_DIR} ${catkin_INCLUDE_DIRS} ) ##################################################
> 
> # Make binary for the offline reader
> ################################################## add_executable(bagconvert src/main.cpp ) target_link_libraries(bagconvert ${Boost_LIBRARIES} ${MATLAB_LIBRARIES_DIR}/libmex.so ${MATLAB_LIBRARIES_DIR}/libmat.so ${MATLAB_LIBRARIES_DIR}/libeng.so ${MATLAB_LIBRARIES_DIR}/libmx.so ${catkin_LIBRARIES})
> 
> Finally, it worked fine! Hope this helps

> it sure worked for me, thanks!

> I delete the findmatlab.cmake and moify the cmakelist as :
> 
> cmake_minimum_required(VERSION 2.8.3)
> 
> # Project name
> project(bagconvert)
> 
> # Find catkin (the ROS build system)
> find_package(catkin REQUIRED COMPONENTS roscpp nav_msgs std_msgs sensor_msgs nav_msgs rosbag)
> 
> # Describe catkin Project
> catkin_package( DEPENDS Eigen3 Boost CATKIN_DEPENDS roscpp nav_msgs std_msgs sensor_msgs nav_msgs rosbag INCLUDE_DIRS src )
> 
> # Include our cmake files
> SET(CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake/)
> 
> # Include libraries
> find_package(Eigen3 REQUIRED) find_package(Boost REQUIRED COMPONENTS system filesystem thread date_time) #find_package(Matlab 3 REQUIRED)
> 
> set(MATLAB_INCLUDE_DIR "/usr/local/MATLAB/R2018a/extern/include") set(MATLAB_LIBRARIES_DIR "/usr/local/MATLAB/R2018a/bin/glnxa64")
> 
> # Include our header files
> include_directories( src ${EIGEN3_INCLUDE_DIR} ${Boost_INCLUDE_DIRS} ${MATLAB_INCLUDE_DIR} ${catkin_INCLUDE_DIRS} ) ##################################################
> 
> # Make binary for the offline reader
> ################################################## add_executable(bagconvert src/main.cpp ) target_link_libraries(bagconvert ${Boost_LIBRARIES} ${MATLAB_LIBRARIES_DIR}/libmex.so ${MATLAB_LIBRARIES_DIR}/libmat.so ${MATLAB_LIBRARIES_DIR}/libeng.so ${MATLAB_LIBRARIES_DIR}/libmx.so ${catkin_LIBRARIES})
> 
> Finally, it worked fine! Hope this helps

>This method workes on 18.04 + Matlab 2021a, LEGEND !!!!!