rpng/kalibr_allan #6: What is mean by IMU remaining still (i.e gravity vector should be alligned with body z-axis)

BharadwajaY, 2018-06-27:
Hi,
I have doubt on capturing the raw imu values while IMU remaining still. That means, we have to keep the IMU z-axis aligned with gravity direction. is this interpretation is correct or we can capture the values at any orientation of IMU with gravity direction.

If we can capture at any orientation, the projection of acceleration due to gravity on xy-plane will be coming as a bias in the measurements. is this interpretation is correct?

goldbattle, 2018-06-28:
The orientation of the IMU should not mater, as this is looking at just the change of the measurements over time. So the assumption is that gravity is constant, thus will remain the same over time. There is not any projection onto the xy-plane, as the bias is computed of each axis independently.

Hope this helps.

goldbattle, 2018-06-28:
And to clarify, when it is said to keep the "IMU still" it means that it should not physically move at all.
This would mean that the IMU is placed where there are no vibrations etc.

BharadwajaY, 2018-06-30:
Thank you for response,
I want one more clarification, Could you help me in finding the scale factor and axis misalignment etc errors of IMU before collecting the data. Since allan variance finds only the stochastics of noise in the data  while the deterministic errors needed to be corrected to the raw data before applying to the allan variance.

goldbattle, 2018-07-01:
For those, as you have stated, this repository will not help.
I would recommend looking in the official kalibr repository, as it should be able to do the IMU misalignment.
https://github.com/ethz-asl/kalibr
http://www.timohinzmann.com/publications/icra_2016_rehder.pdf

BharadwajaY, 2018-07-02:
Thank you for the response and references.

closed 2018-07-02 by BharadwajaY