rpng/kalibr_allan #7: bagconvert segfaults even though it converts to mat successfully

shahvi, 2018-07-04:
Hi,
I keep getting a segfault with:
`*** Error in `../bagconvert/bagconvert': free(): invalid pointer: 0x0000000000e911a8 ***
`
Any ideas what is causing this?  Its not a show stopper but super annoying.

goldbattle, 2018-07-04:
Does this happen when the bag is completely converted and the program is
exiting?
I don't know how to fix it, but my hunch is that it relates to the matlab
library linking/functions.

shahvi, 2018-07-05:
yes i believe that the bag is being converted completely.  I the error happens as the program ends.  If I put an infinite while loop just before the return, I don't see the seg fault.

Possibly something happening in the destructor ?

closed 2019-02-13 by goldbattle