rpng/kalibr_allan #8: Meaning of the line in the graph

imrasp, 2018-07-16:
[image]

I have some trouble about how to interpret these graph. Can some one explain me what is this graph mean. what is cross red and blue line? or suggest me some keyword to search about this.

goldbattle, 2018-07-17:
So those lines are "fit" to the data of the graph.
I would recommend checking out the kalibr wiki for why we put those lines there:
https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model#from-the-allan-standard-deviation-ad

You can also take a look at this gyroscope standard, key page should be 71 (allan variance, section C):
https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=660628&isnumber=14457&tag=1

goldbattle, 2018-07-17:
[image]

closed 2019-02-13 by goldbattle