rpng/kalibr_allan #11: bagconvert failed

eepfs, 2019-03-15:
I used bagconvert to convert a rosbag to a mat. The size of the rosbag is 1.5GB but the size of the mat is 177Bytes. And the mat is a 0*7 array. How do I solve the problem?

goldbattle, 2019-03-18:
I am not sure what the trouble is, but does it report the correct amount of IMU messages that matches the `rosbag info` response? You can also try with some of the demo/example bags as see if it is just your bag that is the issue:
https://drive.google.com/drive/folders/1a3Es85JDKl7tSpVWEUZryOwtsXB8793o?usp=sharing

Let me know if you get it working.

closed 2019-04-03 by goldbattle

0Jiahao, 2019-07-08:
for someone having the similar problem as @eepfs, try using `imu` instead of `\imu`.

theophilec, 2019-07-11:
Hi @goldbattle and @0Jiahao, I am trying to use bagconvert on the `lmu_tango` example data you provided. I have the same problem as @eepfs whether using `imu` or `/imu`.

Any ideas?

Here's an illustration. I renamed the `imu_tango.bag` as `imu.bag`.
<pre><font color="#8AE234"><b>TCantelobre@1044617</b></font>:<font color="#729FCF"><b>~/kvh-calibration</b></font>$ rosrun bagconvert bagconvert data/imu.bag /imu0
[ INFO] [1562871795.376276437]: Starting up
[ INFO] [1562871795.750453405]: BAG Path is: data/imu.bag
[ INFO] [1562871795.750487324]: MAT Path is: data/imu.mat
[ INFO] [1562871795.750491502]: Reading in rosbag file...
[ INFO] [1562871796.385375914]: Done processing bag
[ INFO] [1562871796.385647847]: Done processing IMU data
<font color="#8AE234"><b>TCantelobre@1044617</b></font>:<font color="#729FCF"><b>~/kvh-calibration</b></font>$ ll data
total 519812
drwxr-xr-x 2 TCantelobre TCantelobre      4096 Jul 11 15:03 <font color="#729FCF"><b>.</b></font>/
drwxr-xr-x 6 TCantelobre TCantelobre      4096 Jul 11 14:45 <font color="#729FCF"><b>..</b></font>/
-rw-r--r-- 1 TCantelobre TCantelobre 532269928 Jul 11 14:45 imu.bag
-rw-r--r-- 1 TCantelobre TCantelobre       177 Jul 11 15:03 imu.mat
</pre>
and after removing the i`imu.mat` file,

<pre><font color="#8AE234"><b>TCantelobre@1044617</b></font>:<font color="#729FCF"><b>~/kvh-calibration</b></font>$ rosrun bagconvert bagconvert data/imu.bag imu
[ INFO] [1562871870.084101224]: Starting up
[ INFO] [1562871870.462105810]: BAG Path is: data/imu.bag
[ INFO] [1562871870.462140254]: MAT Path is: data/imu.mat
[ INFO] [1562871870.462144338]: Reading in rosbag file...
[ INFO] [1562871871.098100972]: Done processing bag
[ INFO] [1562871871.098401668]: Done processing IMU data
<font color="#8AE234"><b>TCantelobre@1044617</b></font>:<font color="#729FCF"><b>~/kvh-calibration</b></font>$ ll data
total 519812
drwxr-xr-x 2 TCantelobre TCantelobre      4096 Jul 11 15:04 <font color="#729FCF"><b>.</b></font>/
drwxr-xr-x 6 TCantelobre TCantelobre      4096 Jul 11 14:45 <font color="#729FCF"><b>..</b></font>/
-rw-r--r-- 1 TCantelobre TCantelobre 532269928 Jul 11 14:45 imu.bag
-rw-r--r-- 1 TCantelobre TCantelobre       177 Jul 11 15:04 imu.mat
</pre>

Tal-seven, 2020-06-19:
@theophilec were you able to solve this problem? I'm running into the exact same issue

Tal-seven, 2020-06-20:
i have solved the issue. It was a minor issue, i was specifying the imu topic command line variable wrong. It should have been /uavcanRosBridge/uavcan_ros_bridge/imu as saved in the rosbag file rather than /imu. 
`   if (s1 != NULL && m.getTopic() == imuTopic)`
This check was failing and the dataIMU vector wasnt being populated.