rpng/kalibr_allan #12: gyroscope_random_walk  is NaN

YanhaoZhang, 2019-04-02:
Thank yo so much to update your code. This code works good except the result of gyroscope_random_walk is NaN. May I ask you the reason of it or any hint? I used xsens mti-100 to collect data for 2 hours. Here is my result. Thank you so much. 
[image]

goldbattle, 2019-04-02:
How does the plot look?
It might be that it can't fit the line to the right half of the plot.
That is what I would investigate first.

YanhaoZhang, 2019-04-03:
Many thanks for your prompt reply. You are right the plot does not show the fit line for gyroscope_random_walk. Here is the result. 
[image]

I checked the allan function (by doing '_check_data1 = allan(data1,tau);_') it seems all elements in m are 0 (line 72 in function _allan_). I guess this makes D equals to NaN. May I ask you for any advice? 
I also attached the data I used. The mat file was generated by bagconvert. Thank you so much. 
[https://studentutsedu-my.sharepoint.com/:u:/g/personal/12950426_student_uts_edu_au/EWUvRDDGFkdOiBfz9E8wziIBrPG-6Hb8dltZ20l_DxJivw?e=lZpFDH](url)

> How does the plot look?
> It might be that it can't fit the line to the right half of the plot.
> That is what I would investigate first.

goldbattle, 2019-04-03:
If i remember correctly, the script tries to find the minimum of the curve,
and the left part of the plot should be your white noise, while the right
is your random walk.
I assume it is finding that large spike at the end, so you might need to
cut off your data or figure out why it is having that large spike.
https://github.com/rpng/kalibr_allan/blob/master/matlab/functions/gen_chart.m#L50-L53

It will try to fit a positive 1/2 slope, so the NaN in the y-intercept
hints that it isn't able to fit that, or doesn't have data to do it?

goldbattle, 2019-04-03:
I would try setting the minid to something that you know, for example the halfway of the tau vector size could be a starting point. Hopefully that helps. Checkout this issue for why we fit the lines:
https://github.com/rpng/kalibr_allan/issues/8

YanhaoZhang, 2019-04-03:
Many thanks for your help. I will try all your suggestions :)

closed 2019-06-21 by goldbattle

wwtinwhu, 2019-10-30:
> Many thanks for your prompt reply. You are right the plot does not show the fit line for gyroscope_random_walk. Here is the result.
> [image]
>
> I checked the allan function (by doing '_check_data1 = [...]

Hello, I have the same problem as you. Do you figure out？Thank you very much!

goldbattle, 2019-10-30:
See my comment here: https://github.com/rpng/kalibr_allan/issues/12#issuecomment-479292727
Since you have those large spikes at the end the code will not correctly find the minimum of the curve.
You can try hard coding it to look at everything after 10^1 for example.

wwtinwhu, 2019-10-31:
> See my comment here: [#12 (comment)](https://github.com/rpng/kalibr_allan/issues/12#issuecomment-479292727)
> Since you have those large spikes at the end the code will not correctly find the minimum of the curve.
> You can try hard coding it to look at everything after 10^1 for example.

hello, thank for your reply. I have a question about the code there 
https://github.com/rpng/kalibr_allan/blob/master/matlab/functions/gen_chart.m#L50-L53,  I think it should not find the min of sigavg, but find the min of the difference between the slope(-1/2) and cuver slope. You can check this website by matlab which is different from you as I say below. https://www.mathworks.com/help/nav/ug/inertial-sensor-noise-analysis-using-allan-variance.html#d117e525. This is worth discussing. Thanks.