rpng/kalibr_allan #14: caluate time is too long

Toothsmile, 2019-05-17:
I used this program just to caluate allan var  spend 30 mins, but cant work out ,i dont why need so long time

tuandle, 2019-06-13:
@goldbattle Thanks for sharing the program. 
I have the same problem. I try to reproduce your `imu_visensor` result. I let the matlab program runs for 2 hours but it still doesn't finish. 
Any idea what might cause the problem or is it expected to be super heavy computational load?

araujokth, 2019-06-13:
I am having the same issue. Is this expected? I have ran now for 1 hour and no result yet. I will leave this on during the night and see if anything happens. Also trying with the imu_visensor example

tuandle, 2019-06-14:
It seems so. I reinstalled my Matlab and was able to recompute the `imu_visensor` example in ~30 mins.

goldbattle, 2019-06-21:
Glad the reinstall worked a bit better for you.
I think re-implementing in C++ would be the best idea.
Hopefully down the line will get some time to do that.

closed 2019-06-21 by goldbattle