rpng/kalibr_allan #17: bagconvert failed after Starting up

wwtinwhu, 2019-10-22:
Hello,  I met a error which very fusing.  I have already compiled the bagconvert.  Then I ran the code below
rosrun bagconvert bagconvert imu.bag  /imu
and the output in the terminal is under 
[ INFO] [1571713279.117130547]: Starting up
段错误 (核心已转储)
so I use clion to debug, I find the program breaks at the line below
ros::init(argc, argv, "bagconvert")
It is so confusing. Any Advice? Thank you very much!

goldbattle, 2019-10-22:
This might be a bit tricky to solve and could be caused by a lot of things.
What version of matlab do you link to, and does it run if you do not pass the bag to it?

ZKangsen, 2019-10-25:
hello,
what's the minimum version of matlab?  Is the matlab2016a ok?

goldbattle, 2019-10-25:
I have used this with R2016b before, have not tested with that specific
version, but it is likely to work since we don't use any specific functions.

ZKangsen, 2019-10-25:
> I have used this with R2016b before, have not tested with that specific version, but it is likely to work since we don't use any specific functions.
> […](#)

Got it, but there is another problem. I use the command: rosrun bagconvert bagconvert imu.bag /imu0, will generate empty imu.mat file, but the command: rosrun bagconvert bagconvert imu.bag imu, will be ok.

wwtinwhu, 2019-10-27:
> This might be a bit tricky to solve and could be caused by a lot of things.
> What version of matlab do you link to, and does it run if you do not pass the bag to it?

Hello, The Matlab version is 2017a. The result if  I run bagconvert and not pass a bag to it as follows,
[ INFO] [1572160424.494850947]: Starting up
[ERROR] [1572160424.494896719]: Error please specify a rosbag file
[ERROR] [1572160424.494907735]: Command Example: rosrun bagconvert bagconvert <rosbag> <topic>
free(): invalid pointer
已放弃 (核心已转储)
I plan to change the Matlab version to 2018,  which may fix this problem.

ZKangsen, 2019-10-28:
> > This might be a bit tricky to solve and could be caused by a lot of things.
> > What version of matlab do you link to, and does it run if you do not pass the bag to it?
> 
> @goldbattle Hello, The Matlab version is 2017a. The result if I run bagconvert and not pass a bag to it as follows,
> [ INFO] [1572160424.494850947]: Starting up
> [ERROR] [1572160424.494896719]: Error please specify a rosbag file
> [ERROR] [1572160424.494907735]: Command Example: rosrun bagconvert bagconvert 
> free(): invalid pointer
> 已放弃 (核心已转储)
> I plan to change the Matlab version to 2018, which may fix this problem.

Hey, guy, the problem is not caused by matlab version, it's your incorrect command, you should use it as follow:  rosrun bagconvert bagconvert imu,bag imu, the imu.bag is your imu sensor data.
                                                                                                                                              good luck!

wwtinwhu, 2019-10-29:
> Hey, guy, the problem is not caused by matlab version, it's your incorrect command, you should use it as follow: rosrun bagconvert bagconvert imu,bag imu, the imu.bag is your imu sensor data.
> good [...]

@ZKangsen @goldbattle   Thanks. I change the Matlab version to 2018 and it works. And also the command is as you say,  "/" should not be added to the topic of imu. Thanks again.

closed 2019-10-29 by wwtinwhu

xuhangxx, 2020-03-04:
I found that when I only "rosrun bagconvert bagconvert imu.bag /imu" or "rosrun bagconvert bagconvert imu.bag imu"  in the folder named "data ", it works.