rpng/kalibr_allan #22: should gravity exclude?

suho0515, 2020-04-22:
hello sir, 
i wanna try kalivr_allan with my own IMU sensor. 
and i curious about gravity.

my sensor output include gavity value. like the example below.

acc_x : -0.001
acc_y : -0.003
acc_z : -0.984

gyr_x : 0.004
gyr_y : -0.001
gyr_z : -0.004

is it okay to have gravity value in my acceleration?
if it's not okay, i'm gonna fix the value.

and thanks for your wonderful work.

regards.

goldbattle, 2020-04-22:
We are looking at the allan *deviation* and since the sensor does not move
this shouldn't be a problem.
Your gravity also seems to be a bit small.
Typically one would expect the norm of a_m to be around 9.81.

suho0515, 2020-04-23:
> We are looking at the allan *deviation* and since the sensor does not move this shouldn't be a problem. Your gravity also seems to be a bit small. Typically one would expect the norm of a_m to be around 9.81.
> […](#)

i understood, thanks for your kindness. 

thanks again! :)

closed 2020-04-24 by goldbattle