rpng/kalibr_allan #23: How to transfer IMU data to a rosbag?

Nronaldo, 2020-05-15:
hello, I appreciate your nice work.
I am a newbie to ROS. I download your android-dataset-recorder and record images and IMU data.
I wonder how can I transfer the IMU data to a rosbag? I know there are some tools to create rosbag  with both IMU data and images. But I don not know how to create a rosbag with only IMU. Could you please give me some advice?

goldbattle, 2020-05-15:
I think there was a PR which handled reading from a text file.
https://github.com/rpng/kalibr_allan/pull/16
Else this answer might help you:
https://stackoverflow.com/a/53923403/7718197

Sandeldeng, 2020-07-16:
rosrun bagconvert bagconvert imu.bag /imu0

NickPerezCarletonUniversity, 2021-09-20:
what are all the different fields that someone can populate for the Imu() object? I can't find the documentation or code for it.