rpng/kalibr_allan #28: Package availability for ROS1 Noetic

Enes1097, 2021-08-13:
Hey,
I wanted to ask if this package is available for ROS1 Noetic?
Which is the default distribution this package runs on?

petarsecond, 2021-08-30:
This fork is for Ubuntu 20.04 and Noetic dist https://github.com/ori-drs/kalibr

Enes1097, 2021-08-31:
@petarsecond thank you for your answer. But I was asking about the kalibr allan package to compute the imu noise values, not the normal Kalibr package. 
Sadly the ori-drs package doesn't work for me, because it stops computing after the first jacobi matrix output (https://github.com/ethz-asl/kalibr/issues/396#issuecomment-820242933).

valenbase, 2021-09-13:
hi @petarsecond when i use https://github.com/ori-drs/kalibr,there are two problems:
1\  after source setup.zsh cannot find the kalibr_calibrate*** commands
2\ the kalibr cannot create the result file